nRF24/RF24Gateway #1: RF24Gateway_ncurses causes wlan0 if to go down

y8, 2015-04-22:
When I'm starting ``RF24Gateway_ncurses`` on my RPI 2, wlan0 interface immediately goes down:

````
Apr 22 14:09:24 raspberrypi ifplugd(tun_nrf24)[2561]: Link beat detected.
Apr 22 14:09:24 raspberrypi ifplugd(tun_nrf24)[2561]: Executing '/etc/ifplugd/ifplugd.action tun_nrf24 up'.
Apr 22 14:09:24 raspberrypi wpa_supplicant[4257]: wlan0: CTRL-EVENT-DISCONNECTED bssid=00:00:00:00:00:00 reason=0
Apr 22 14:09:24 raspberrypi ifplugd(tun_nrf24)[2561]: client: OK
Apr 22 14:09:24 raspberrypi ifplugd(tun_nrf24)[2561]: client: Ignoring unknown interface tun_nrf24=tun_nrf24.
Apr 22 14:09:24 raspberrypi wpa_action: WPA_IFACE=wlan0 WPA_ACTION=DISCONNECTED
Apr 22 14:09:24 raspberrypi wpa_action: WPA_ID=0 WPA_ID_STR= WPA_CTRL_DIR=/var/run/wpa_supplicant
Apr 22 14:09:24 raspberrypi wpa_action: ifdown wlan0
Apr 22 14:09:24 raspberrypi dhclient: Internet Systems Consortium DHCP Client 4.2.2
Apr 22 14:09:24 raspberrypi dhclient: Copyright 2004-2011 Internet Systems Consortium.
Apr 22 14:09:24 raspberrypi dhclient: All rights reserved.
Apr 22 14:09:24 raspberrypi dhclient: For info, please visit https://www.isc.org/software/dhcp/
Apr 22 14:09:24 raspberrypi dhclient: 
Apr 22 14:09:24 raspberrypi ifplugd(tun_nrf24)[2561]: Program executed successfully.
Apr 22 14:09:24 raspberrypi ifplugd(wlan0)[4240]: Link beat lost.
Apr 22 14:09:24 raspberrypi dhclient: Listening on LPF/wlan0/24:05:0f:xx:xx:xx
Apr 22 14:09:24 raspberrypi dhclient: Sending on   LPF/wlan0/24:05:0f:xx:xx:xx
Apr 22 14:09:24 raspberrypi dhclient: Sending on   Socket/fallback
Apr 22 14:09:24 raspberrypi dhclient: DHCPRELEASE on wlan0 to 192.168.1.254 port 67
Apr 22 14:09:24 raspberrypi wpa_action: removing sendsigs omission pidfile: /run/sendsigs.omit.d/wpasupplicant.wpa_supplicant.wlan0.pid
````

TMRh20, 2015-04-22:
Another user reported a similar problem. His solution was:

*Watching at the daemon.logs i found out that ifplugd was responsible for
this automation.So i edited /etc/default/ifplugd like
thisINTERFACES="auto"HOTPLUG_INTERFACES="eth0 wlan0"*

closed 2015-11-06 by TMRh20

myTonino, 2016-08-17:
I have the very same problem on a RPi2 under raspian. A pity as I wanted to follow that beautiful guide http://tmrh20.blogspot.de/2016/08/raspberry-pilinux-with-nrf24l01.html
using the RF24Gateway to monitor my RF24mesh using wireshark. I am running all the latest raspian and NRF24 software.

I restricted ifplugd defaults as above to
`"HOTPLUG_INTERFACES="eth0 wlan0"`

But in my case both interfaces, eth0 and wlan0, loose contact, although there is no trace anymore of ifplugd in daemon.logs.

A simple
```
# sudo ifdown wlan0
# sudo ifup --force wlan0
```
does not reactivates them. Only
```
# sudo ifconfig tun_nrf24 down
# sudo ifdown wlan0
# sudo ifup --force wlan0
```
I would appreciate any idea on what to look for.

myTonino, 2016-08-17:
Resolved my issue. The wlan interface was up and running, just the traffic was misrouted. I shouldn't have started the NR24Gateway on the same subnet as my wlan0.... If someone writes some documentation on this setup, one could put that part in BOLD;)

Sorry for the noise. Someone with close to no knowledge on networking shouldn't play around with such cool things.

TMRh20, 2016-08-18:
> Sorry for the noise. Someone with close to no knowledge on networking shouldn't play around with such cool things.

@myTonino Heh well now you have some, and thats what its for!

myTonino, 2016-08-18:
@TMRh20 Thanks for sharing all those cool things and all your expertise. Very appreciated! Keep on going!