nRF24/RF24Gateway #2: Simple Gateway with Arduino/ATmega (question)

crcastle, 2015-07-08:
Is there something like RF24Gateway for ATmega?

I'm prototyping a sensor network that uses ATmega328P MCUs, and I'd like to avoid having to add in a Rasp Pi to bridge sensor data to the internet as (I think) this library requires (for power reasons mostly).  My plan was to have the master/gateway node use both an nRF24L01+ and an ESP8266 module.  I've got a couple sensors running using your RF24 library now, and I'm planning on moving to your RF24Network library to simplify addressing, routing, reliability, maintenance, and growth.

I think it won't be too complex to have the RF24Network master node receive messages from the sensors and generate the appropriate HTTP or MQTT message to send over wifi, but wanted to check if you know of code that does this since it seems like a common use case.  Note that the sensors will never have to receive data from the TCP/IP network.

Also, thanks a ton not only for these libraries, but also for the awesome documentation of them.

TMRh20, 2015-07-09:
Well, yes and no lol?

RF24Gateway is more or less a gateway for both TCP/IP and RF24Network nodes, potentially running a combination of RF24Ethernet, RF24Network & RF24Mesh nodes. If you don't need or want direct TCP/IP capabilities on your RF24 sensors, you can just use an RF24Network or RF24Mesh master node as the gateway. From there, it is a matter of the capabilities of the ESP module. 

I hav seen this: https://github.com/tuanpmt/esp_mqtt , and I believe ESP has a lib similar to the Arduino Ethernet API, but I haven't experimented at all myself with ESP modules.

crcastle, 2015-07-12:
Since the ESP8266 is a pretty powerful MCU, I'm trying to use that as my master/gateway.  I've got simple Arduino programs running on the ESP8266 but am [having problems getting RF24 functionality to work](https://gist.github.com/crcastle/93ebe3b42f3ab021639b).  Although I have a feeling the problems might be related to the watchdog timer on the ESP.  Once I get RF24 functionality working, I'm going to use your RF24Network on the ESP (and on the ATmega-based sensor nodes in the field).

Regardless, thanks for these links.  They will be helpful if I give up on using the ESP as MCU and use an ATmega.

closed 2015-07-12 by crcastle