nRF24/RF24Gateway #3: Send/Receive Loop when trying to use MQTT over RF24Ethernet/RF24Gateway

Krifto, 2015-09-05:
When putting MQTT client on an Arduino UNO, thanks to the awesome RF24* libraries (which implement the Client.h interface) we can have it communicate over the RF24Ethernet library to a mosquitto server on a Raspberry Pi. The MQTT client is found here https://github.com/knolleary/pubsubclient, the MQTT server is mosquitto.

Everything works great (publishing and subscribing messages, reconnecting etc) but I have found a strange problem. I noticed lots of TX and RX traffic on the RF24Gateway dashboard. Running the mosquitto server with verbose output (mosquitto -v), I could see that the Arduino was constantly firing a SUBSCRIBE request, and the mosquitto server was acknowledging it. 

Racking up the RF24Ethernet debugging, I can see that the code repeatedly reaches this line in RF24Client.cpp:

`IF_ETH_DEBUG_L2( Serial.print(F("UIPClient stop?????? ")); Serial.println(data->state & UIP_CLIENT_SOCKETS); Serial.println(F("**")); Serial.println(data->state,BIN); Serial.println(F("**")); Serial.println(UIP_CLIENT_SOCKETS,BIN); Serial.println(F("**"));  );`.

This produces the output `UIPClient stop?????? 0
    **
    10000
    **
    11111111111111111111111100001111
    **`.
		
After a while, this back-and-forth ends, but then the same starts happening when there is a ping handshake between client and server. Then the whole RF2Ethernet connection seems to drop, and after reconnect everything starts over again.

I have tried all kinds of configuration changes, tried to trace through all the different libraries (including PubSubClient). I noticed there that the PubSubClient actually only sends the SUBSCRIBE request once, but somehow RF24* repeats sending that same request, every time the SUBACK message from the MQTT server arrives.

I feel that I do not understand the whole stack of libraries enough to make any headway with this problem. 

Could somebody else maybe try the sketch and see if they encounter the same problem? Or maybe someone already got MQTT over RF24 running smoothly?

Here is the sketch I am trying:
https://gist.github.com/Krifto/865456a1f9596c0878b3

TMRh20, 2015-09-17:
FYI, I had a chance to test this a bit and did encounter the same problem using the sketch. 

I noticed there is a different library for the ENC28J60 chip, which uses the same UIP IP stack as RF24Ethernet.

Per the pubsubclient repository:
> The library cannot currently be used with hardware based on the ENC28J60 chip – such as the Nanode or the Nuelectronics Ethernet Shield. For those, there is an alternative library available.

So I am thinking that either some changes are probably required to make this work, or it might require direct use of the UIP stack, connected to RF24Network. 

In a quick run-through, I simply removed the 'http client hack' and it seemed to work up to a point, but there seems to be a problem with things not being acked correctly or something. If interested I can post my changes so far to RF24Ethernet, but I would need a bit of time to clean them up etc.

TMRh20, 2015-09-17:
Forgot to add, here is the output from mosquitto after my latest changes:

```
 New connection from 10.10.3.9 on port 1883.
1442431454: New client connected from 10.10.3.9 as arduinoClient (c1, k10).
1442431454: Sending CONNACK to arduinoClient (0, 0)
1442431454: Received PUBLISH from arduinoClient (d0, q0, r0, m0, 'outTopic', ... (11 bytes))
1442431454: Received SUBSCRIBE from arduinoClient
1442431454:     inTopic (QoS 0)
1442431454: arduinoClient 0 inTopic
1442431454: Sending SUBACK to arduinoClient
1442431459: Received PINGREQ from arduinoClient
1442431459: Sending PINGRESP to arduinoClient
1442431459: Received PINGREQ from arduinoClient
1442431459: Sending PINGRESP to arduinoClient
1442431463: Received PINGREQ from arduinoClient
1442431463: Sending PINGRESP to arduinoClient
1442431478: Client arduinoClient has exceeded timeout, disconnecting.
1442431478: Socket error on client arduinoClient, disconnecting.

```

TMRh20, 2015-11-29:
Been a while, but now fixed in the latest update & added an example: http://tmrh20.github.io/RF24Ethernet/mqtt_basic_8ino-example.html

Krifto, 2015-11-30:
Oh wow, that sounds great! Sorry for not replying to your earlier post, I did not have the time to look into this again. I will try to check your updated version soon, thanks!

Krifto, 2015-12-05:
I can confirm that this works, thanks a lot! The combination of RF24Ethernet/RF24Gateway and MQTT makes Arduino+nrf24l01 a very nice alternative to solutions based on the Esp8266 Wifi chips.

One thing I noted: In the example code, it is better to get rid of the `while` loop in the `reconnect()` function:

    while (!client.connected())...

If the mesh disconnects for some reason, the code would remain stuck in this `while` loop forever.

closed 2015-12-05 by Krifto

psprajc, 2015-12-11:
This is great news, that mqtt is working ower rf24 radio. Thanks for great work again to TMRh20.

I will try this new solution. Previous version did hang in some loop.

Would it be possible to connect over mqtt-sn? Because mqtt is heavier in overhead and also works over TCP?

What do you think?

TMRh20, 2015-12-11:
> Previous version did hang in some loop.

The git version of RF24Ethernet incorporates a ``` yield() ``` function I noticed in Arduino [wiring.c](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/wiring.c)  that allows the IP stack to be refreshed while ``` delay() ``` is called or the program is waiting. I found that without it, the mqtt example may hang. I've been meaning to do a bit more work and put out the changes in a release, but haven't had a chance.

>Would it be possible to connect over mqtt-sn? Because mqtt is heavier in overhead and also works over TCP?

Yes? I think if a lightweight protocol is the idea, it might be best to look at incorporating mqtt-sn in with RF24Network/RF24Mesh because IP & RF24Ethernet bring a lot of overhead by comparision. I think enabling UDP in RF24Ethernet also increases resource usage, but there might be ways around that.