nRF24/RF24Gateway #5: 'rfNoInterrupts' was not declared in this scope

yozik04, 2015-12-22:
Can not compile.
```
RF24Gateway$ make
g++ -Wall -fPIC -std=c++0x -Ofast -mfpu=vfp -mfloat-abi=hard -march=armv7-a -mtune=arm1176jzf-s  -c RF24Gateway.cpp
RF24Gateway.cpp: In member function 'void RF24Gateway::poll(uint32_t)':
RF24Gateway.cpp:262:20: error: 'rfNoInterrupts' was not declared in this scope
     rfNoInterrupts();
                    ^
RF24Gateway.cpp:264:18: error: 'rfInterrupts' was not declared in this scope
     rfInterrupts();
                  ^
RF24Gateway.cpp: In member function 'void RF24Gateway::handleRX(uint32_t)':
RF24Gateway.cpp:441:28: error: 'rfNoInterrupts' was not declared in this scope
             rfNoInterrupts();
                            ^
RF24Gateway.cpp:450:26: error: 'rfInterrupts' was not declared in this scope
             rfInterrupts();
                          ^
RF24Gateway.cpp: In member function 'void RF24Gateway::handleTX()':
RF24Gateway.cpp:466:24: error: 'rfInterrupts' was not declared in this scope
           rfInterrupts();
                        ^
RF24Gateway.cpp:472:26: error: 'rfInterrupts' was not declared in this scope
             rfInterrupts();
                          ^
Makefile:49: recipe for target 'RF24Gateway.o' failed
make: *** [RF24Gateway.o] Error 1
```

Maybe a file is missing?

TMRh20, 2015-12-23:
This error is due to some recent development on the RF24 core lib and RF24Gateway to include interrupts, and bring CPU usage down to a manageable level. In other words, the RF24 library just needs to be updated. There is an install and updater script [here](https://github.com/TMRh20/tmrh20.github.io/tree/master/RF24Installer/RPi) if you are into that kind of thing. :p

See https://github.com/TMRh20/RF24/issues/172 for more info on interrupt functionality.

yozik04, 2015-12-23:
Thank you. I was able to compile it now.
Created some pull requests to autodetect RPi2.

closed 2015-12-23 by yozik04