nRF24/RF24Gateway #7: Examples: ncursesInt does not accept connections

yozik04, 2015-12-23:
ncurses with interrupts does not accept connections.

But plain ncurses works fine except for 100% CPU usage. I connect to MQTT and successfully send and receive messages.

Am really not sure how to debug...
I run both with root rights.

I use Arduino Nano with RF24 module as client.
How RPi should get interrupt from RF module any additional wiring required?

TMRh20, 2015-12-24:
Ahh, I think so. Interrupts do require the [wiringPi GPIO utility](https://projects.drogon.net/raspberry-pi/wiringpi/the-gpio-utility/) which seems to be included in the current raspbian. If installed, you should get some info calling ```gpio -v``` 

Per the pinout at the bottom of the [rf24 docs](http://tmrh20.github.io/RF24/) the spare IRQ pin can connect to your chosen RPi GPIO pin (default 23).

Then in the .cpp file(s) ``` attachInterrupt(23, INT_EDGE_FALLING, intHandler); ```  23 is the GPIO number which can be modified as you choose.

This is still being put together but we will get some of this info documented soon.

yozik04, 2015-12-24:
Thank you for help. I managed to get it working. IRQ pin was not connected.
I had to install wiringPi as well using this manual http://wiringpi.com/download-and-install/. I am using OSMC on my Pi2. It does not have wiringPi preinstalled and does not have it in available packages.

closed 2015-12-24 by yozik04