nRF24/RF24Gateway #8: Packets that are 117 bytes or more get corrupted

conmarap, 2018-09-30:
I have a very particular issue. I compiled the project and installed it on two Raspberry Pis. Then I compiled [this](https://github.com/nRF24/RF24Gateway/blob/master/examples/RF24GatewayNode.cpp) example and ran it on the two Raspberry Pis.

I was able to ping each device back and forth.

``` sh
ping 10.10.2.1 # From 10.10.2.2
ping 10.10.2.2 # From 10.10.2.1
```

Then, naturally, I tried to `curl` one of the devices from the other, but nothing happened. I verified that packets were being sent with `tcpdump`. The one weird thing that I noticed was that even though the packets are being received, my `curl 10.10.2.2` did not show a result, which may be because of the repeating warning that `326 bytes missing!`.

``` sh
sudo tcpdump -i tun_nrf24
```
```
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on tun_nrf24, link-type RAW (Raw IP), capture size 262144 bytes
20:51:37.131645 IP 10.10.2.1.48884 > 10.10.2.2.http: Flags [S], seq 131757859, win 29200, options [mss 1460,sackOK,TS val 2462727011 ecr 0,nop,wscale 7], length 0
20:51:37.143502 IP 10.10.2.2.http > 10.10.2.1.48884: Flags [S.], seq 1415731283, ack 131757860, win 28960, options [mss 1460,sackOK,TS val 4243499299 ecr 2462727011,nop,wscale 6], length 0
20:51:37.143724 IP 10.10.2.1.48884 > 10.10.2.2.http: Flags [.], ack 1, win 229, options [nop,nop,TS val 2462727023 ecr 4243499299], length 0
20:51:37.144036 IP 10.10.2.1.48884 > 10.10.2.2.http: Flags [P.], seq 1:74, ack 1, win 229, options [nop,nop,TS val 2462727023 ecr 4243499299], length 73: HTTP: GET / HTTP/1.1
20:51:37.166563 IP 10.10.2.2.http > 10.10.2.1.48884: Flags [.], ack 74, win 453, options [nop,nop,TS val 4243499322 ecr 2462727023], length 0
20:51:37.358333 IP truncated-ip - 326 bytes missing! 10.10.2.2.http > 10.10.2.1.48884: Flags [P.], seq 1:419, ack 74, win 453, options [nop,nop,TS val 4243499508 ecr 2462727023], length 418: HTTP: HTTP/1.1 401 Unauthorized
20:51:37.592104 IP truncated-ip - 326 bytes missing! 10.10.2.2.http > 10.10.2.1.48884: Flags [P.], seq 1:419, ack 74, win 453, options [nop,nop,TS val 4243499736 ecr 2462727023], length 418: HTTP: HTTP/1.1 401 Unauthorized
20:51:37.817547 IP truncated-ip - 326 bytes missing! 10.10.2.2.http > 10.10.2.1.48884: Flags [P.], seq 1:419, ack 74, win 453, options [nop,nop,TS val 4243499966 ecr 2462727023], length 418: HTTP: HTTP/1.1 401 Unauthorized
20:51:38.266302 IP truncated-ip - 326 bytes missing! 10.10.2.2.http > 10.10.2.1.48884: Flags [P.], seq 1:419, ack 74, win 453, options [nop,nop,TS val 4243500416 ecr 2462727023], length 418: HTTP: HTTP/1.1 401 Unauthorized
20:51:39.218605 IP truncated-ip - 326 bytes missing! 10.10.2.2.http > 10.10.2.1.48884: Flags [P.], seq 1:419, ack 74, win 453, options [nop,nop,TS val 4243501367 ecr 2462727023], length 418: HTTP: HTTP/1.1 401 Unauthorized
20:51:41.058371 IP truncated-ip - 326 bytes missing! 10.10.2.2.http > 10.10.2.1.48884: Flags [P.], seq 1:419, ack 74, win 453, options [nop,nop,TS val 4243503207 ecr 2462727023], length 418: HTTP: HTTP/1.1 401 Unauthorized
20:51:43.372121 IP 10.10.2.2.http > 10.10.2.1.48884: Flags [F.], seq 419, ack 74, win 453, options [nop,nop,TS val 4243505529 ecr 2462727023], length 0
20:51:43.372326 IP 10.10.2.1.48884 > 10.10.2.2.http: Flags [.], ack 1, win 229, options [nop,nop,TS val 2462733251 ecr 4243499322,nop,nop,sack 1 {419:420}], length 0
20:51:43.389478 IP truncated-ip - 326 bytes missing! 10.10.2.2.http > 10.10.2.1.48884: Flags [P.], seq 1:419, ack 74, win 453, options [nop,nop,TS val 4243505539 ecr 2462733251], length 418: HTTP: HTTP/1.1 401 Unauthorized
20:51:43.617222 IP truncated-ip - 326 bytes missing! 10.10.2.2.http > 10.10.2.1.48884: Flags [P.], seq 1:419, ack 74, win 453, options [nop,nop,TS val 4243505767 ecr 2462733251], length 418: HTTP: HTTP/1.1 401 Unauthorized
20:51:44.068513 IP truncated-ip - 326 bytes missing! 10.10.2.2.http > 10.10.2.1.48884: Flags [P.], seq 1:419, ack 74, win 453, options [nop,nop,TS val 4243506217 ecr 2462733251], length 418: HTTP: HTTP/1.1 401 Unauthorized
20:51:44.977209 IP truncated-ip - 326 bytes missing! 10.10.2.2.http > 10.10.2.1.48884: Flags [P.], seq 1:419, ack 74, win 453, options [nop,nop,TS val 4243507127 ecr 2462733251], length 418: HTTP: HTTP/1.1 401 Unauthorized
20:51:46.817625 IP truncated-ip - 326 bytes missing! 10.10.2.2.http > 10.10.2.1.48884: Flags [P.], seq 1:419, ack 74, win 453, options [nop,nop,TS val 4243508967 ecr 2462733251], length 418: HTTP: HTTP/1.1 401 Unauthorized
20:51:50.417021 IP truncated-ip - 326 bytes missing! 10.10.2.2.http > 10.10.2.1.48884: Flags [P.], seq 1:419, ack 74, win 453, options [nop,nop,TS val 4243512567 ecr 2462733251], length 418: HTTP: HTTP/1.1 401 Unauthorized
20:51:57.617000 IP truncated-ip - 326 bytes missing! 10.10.2.2.http > 10.10.2.1.48884: Flags [P.], seq 1:419, ack 74, win 453, options [nop,nop,TS val 4243519767 ecr 2462733251], length 418: HTTP: HTTP/1.1 401 Unauthorized
20:52:06.005643 IP 10.10.2.1.48884 > 10.10.2.2.http: Flags [F.], seq 74, ack 1, win 229, options [nop,nop,TS val 2462755885 ecr 4243499322,nop,nop,sack 1 {419:420}], length 0
20:52:06.016410 IP 10.10.2.2.http > 10.10.2.1.48884: Flags [.], ack 75, win 453, options [nop,nop,TS val 4243528173 ecr 2462755885], length 0
20:52:12.342754 IP truncated-ip - 326 bytes missing! 10.10.2.2.http > 10.10.2.1.48884: Flags [P.], seq 1:419, ack 75, win 453, options [nop,nop,TS val 4243534487 ecr 2462755885], length 418: HTTP: HTTP/1.1 401 Unauthorized
```

I opened up port `6969` with netcat:

``` sh
while true; do
    nc -l 6969
done
```

on one of the Pis and on the other I started sending characters to see where the cutoff was:

``` sh
for i in `seq 20 120`; do
    python -c "print 'A' * $i" | nc -q0 10.10.2.2 6969;
    sleep 1;
done
```

I noticed that all data was sent up to 116 bytes. Thereafter the packets were dropped. I couldn't see anything that's preventing the packets from being sent, especially since I see the tcpdump log on the receiving end, but I'm pretty sure the warning that `326 bytes missing!` is what's to blame.

Is there a way to patch this issue?

conmarap, 2018-09-30:
Thank you! I didn't think to look in the network layer and must've missed it in the documentation.

Will this prevent a mix of Pis and Arduino on sharing the same network?

TMRh20, 2018-09-30:
Nope, Arduinos can still function along with RPis they are just limited in payload sizes etc

conmarap, 2018-09-30:
Thank you again!

closed 2018-09-30 by conmarap