nRF24/RF24Gateway #9: Two Raspberry Pis as a bridge

donei34, 2019-09-14:
Hi,

Firstly, thank you very much for your amazing libraries! 

I was wondering how I would go about using two gateways together for the purposes of creating a long distance internet link between two raspberry pi's?

So far, I have installed every library and managed to compile the ncurses example, however I am unsure what I need to change in order for the two gateways to have different addresses and connect to one another. The ncurses example works on each Pi, but no connection is shown, and I cannot ping the other Pi through the tun adapter.

Thanks!

TMRh20, 2019-09-15:
Hi, 

As long as you assign one of the devices a non-zero nodeID, and a matching IP address (last octet ==nodeID), they should be able to connect. 

I’ve had some stability issues with RF24Gateway/RPi which I’ve not been able to fix, so I’d be interested in how well it works for you.

Peace,
TMRh20

donei34, 2019-09-15:
Hi,

So since you mentioned stability I decided to strip my testing back to purely the RF24 base library. My setup is with one Pi 3B and one Pi 3B+. I tested the modules with the getting started and pingpair programs, but encountered some issues:

- The Pi 3B+ seems to be able to transmit and receive data
- The Pi 3B seems to be able to receive data but not transmit data

I have tried rewiring the Pi 3B module connections, swapping the modules between the two Pis, and swapping the sdcards between the two Pis - nothing has helped.

The only differences I can currently see between the two are:

- The fact that one Pi is a 3B
- My power supply for each Pi is different, and likely to only provide 1A or 1.5A

Breaking down the 3B vs 3B+ further:

- I changed the wires on the 3B
- I checked the wiring on the 3B - it is the same as on the 3B+
- I made sure that all wires were firmly connected to the gpio header

Any ideas as to what could be causing the Pi 3B to not be able to transmit data?

Thanks

TMRh20, 2019-09-15:
Good move. I would guess that the problem comes down to the radios themselves. Since you have a one way communication I’m guessing one radio just can’t “hear” the other. If you have the modules with pcb antenna, I would suggest scraping down a bit of the antenna trace and soldering on a 3 to 4 inch piece of wire to function as a proper antenna.

donei34, 2019-09-15:
By scraping down the antenna, do you mean just removing the end of it, actually cutting the antenna trace or something else? Is there a guide for this?

TMRh20, 2019-09-16:
I mean just scraping the trace clean so you can solder onto it. No guide, just some hackish stuff to get things working :p

donei34, 2019-09-16:
Unfortunately adding the wire antenna didn't help. The weird thing is that it's not either of the modules - the only aspect that seems to be affecting whether the modules can transmit or not is whether it is connected to the Pi 3B or not.

TMRh20, 2019-09-16:
How about if you use the SPIDEV driver? Run ./configure —driver=SPIDEV  , then  make; sudo make install   from the Rf24 directory.

donei34, 2019-09-17:
Unfortunately that did not work either, however using two other Pi 3Bs in a friend's setup makes it work. Not quite sure what went wrong in my setup, but it's working for now. Thanks for all of your help!

closed 2019-09-17 by donei34