nRF24/RF24Gateway #10: Is it possible to run the RF24 GATEWAY gateway in the background?

potier97, 2020-04-08:
Is it possible to run the RF24 GATEWAY gateway in the background?

Hello, again, I am using NRFGATEWAY for a raspberry pi 3, which works very well as a coordinating node, but I would like to know if it is possible to run it in the background, when we say the power goes out and the system restarts, I can start RF24GATEWAY again from the ncurses example, and passing it the ip address and netmask automatically, is this possible?

Thank you

TMRh20, 2020-04-09:
Hola, yup it is,

preconfigure IF:
`ip tuntap add dev tun_nrf24 mode tun user pi multi_queue`
`ifconfig tun_nrf24 10.10.2.x/24`

You need [screen](https://linux.die.net/man/1/screen) for the following:
Run with ncurses, where `pi` is the user:
`su - pi -c "screen -S RF24Gw_on_RPi3 -dm sudo /home/pi/rf24libs/RF24Gateway/examples/ncursesInt/./RF24Gateway_ncursesInt&"`

You can put the above into a script of course, to run on boot.

Access the screen: `screen -r`
Detatch the screen: `Ctrl + A then Ctrl + D`

potier97, 2020-04-09:
Hi, I am going to test what you told me and I tell you how it turned out, however, I was thinking and wondering, that, if it is necessary to use NRF24GATEWAY for a network of nodes that communicate with NRF24L01 through the MQTT protocol ?, since This protocol is light and thought of in the IOT, which you can advise me in this case.

Thank you

TMRh20, 2020-04-09:
Well, yes and no?
If you want a direct MQTT connection using nrf24 then yes.
If you use RF24Network or RF24Mesh you can send data to RPi, but need to write RPi code to pass data to/from the MQTT server  and the nodes.

TMRh20, 2020-04-09:
Just randomly thinking about this question, and I should probably add that there are a few other options, mainly if you have an esp8266, it would be easiest to have the esp as RF24Mesh or RF24Network master node, and act as a MQTT "bridge" for the sensor data using WiFi. Depending on what you want to do, you could also check out the MySensors system, which can use nrf24l01 radios.

potier97, 2020-04-09:
Hello again, I am looking at what you told me with the MySensors system, although it has things similar to yours, however, it is not so easy to understand at first, I am trying to connect and make a simple gateway and I have not yet done so achieved, but then would it be better to work with these libraries than with yours?

TMRh20, 2020-04-09:
Well it’s a matter of preference really. MySensors is more of a complete package, so less coding involved, where the RF24 stack requires users to write their own code, so is a bit more flexible.

potier97, 2020-04-09:
> Bueno, es una cuestión de preferencia realmente. MySensors es más un paquete completo, por lo que implica menos codificación, donde la pila RF24 requiere que los usuarios escriban su propio código, por lo que es un poco más flexible.
> [...](#)
> El 9 de abril de 2020, a las 14:12, potier97 ***@***.***> escribió: Hola de nuevo, estoy mirando lo que me dijiste con el sistema MySensors, aunque tiene cosas similares a las tuyas. , sin embargo, al principio no es tan fácil de entender, estoy tratando de conectarme y crear una puerta de enlace simple y aún no lo he logrado, pero ¿sería mejor trabajar con estas bibliotecas que con las suyas? - Estás recibiendo esto porque comentaste. Responda a este correo electrónico directamente, mírelo en GitHub o cancele la suscripción.

I understand, since my purpose is to be able to send this data to the AWS cloud using nrf24 and the mqtt protocol, I see that Missensores has not yet managed to do this within their development, and they handle processes, I still do not understand why they are needed ? , but I am trying to do a little test to see if it is possible, I don't need a graphical interface to take my data to the cloud, but I can treat the data that is sent to the cloud so that it really sends useful information, and not junk data that fill a database and generate cost overruns

What would you recommend for this solution? I'm still very new to these things

TMRh20, 2020-04-09:
> What would you recommend for this solution? I'm still very new to these things

As I mentioned before, I would suggest using the ESP8266 as the master node, running RF24Mesh for the RF24 nodes. The ESP8266 would collect the raw data, format it correctly, then send it out using MQTT etc.
There are well written libraries available to do this type of thing on ESP devices, so that would be the easiest method IMHO.

potier97, 2020-04-10:
> > What would you recommend for this solution? I'm still very new to these things
> 
> As I mentioned before, I would suggest using the ESP8266 as the master node, running RF24Mesh for the RF24 nodes. The ESP8266 would collect the raw data, format it correctly, then send it out using MQTT over WiFi.

I was thinking about your answer, I want to use a raspberri for my gateway solution, in addition to the requirements that I planted at the beginning is that I need to implement a sensor network where each node can communicate with mqtt, so the bridge could not be an option, test your mqtt example with RF24Gateway and it works perfect (although there are things that I still don't understand) in a test environment and low load, I would like to deploy it in a network of initially 9 nodes, and since I want the network to be more The gateway would become a bottleneck if I left it as a bridge, recently I saw node-red, to implement event programming, and it can meet my requirements to send data to the Amazon cloud, when the data changes according to the algorithm you build, could you make this possible with your ncurses example?

TMRh20, 2020-04-10:
> The gateway would become a bottleneck if I left it as a bridge

I very much doubt that. RF24Ethernet has a very high overhead by comparison

> could you make this possible with your ncurses example

No

potier97, 2020-04-10:
ok, i am going to keep trying to make this configuration to generate a more robust solution, likewise i would like to contribute with your project, since i quite like the idea of managing these devices. And that execution in the background works correctly for the Gateway.

Thanks again

closed 2020-07-26 by potier97