nRF24/RF24Gateway #12: RF24Gateway master node can't route external packets to other nodes

SimonMerrett, 2020-07-08:
Hi, as discussed in emails, it seems that the Pi running the RF24Gateway master node (0) can't route packets destined to be delivered via other nodes if it has received them from outside the Pi.

A rough workaround to deliver specific packets from master node to a single other node (33) was tested and works:

```
//if ( (meshAddr = mesh.getAddress(lastOctet)) > 0 || thisNodeID) {
if ( (meshAddr = mesh.getAddress(33)) > 0 || thisNodeID) {  
  RF24NetworkHeader header(meshAddr, EXTERNAL_DATA_TYPE);
  if(thisNodeID){ //If not the master node, send to master (00)
    header.to_node = 00;
  }
  ok = network.write(header, msgTx->message, msgTx->size);
```
A patch is requested which filters destination and own IP/mask and sends to specific nodes in response.

Thanks for the libraries!

TMRh20, 2020-07-09:
Ok, I think I've got this figured out, changed my network config to align, so needs a bit of an explanation I guess.

**Network Configuration:**
Local WiFi/Ethernet Network (Windows,Linux,RPi devices etc) - `10.10.1.0 - 255.255.255.0`  ( **10.10.1.x network** )
RPi devices (tun_nrf24 interfaces & all connected devices) - `10.1.1.0 - 255.255.0.0` ( **10.1.x.x network** ) 

Explanation: The RF24Gateway/RF24Mesh network supports about 253 hosts. I setup the RF24Gateway and all associated devices with a much larger subnet in this config to encompass all devices directly accessible via the RF24Gateway network.

ie: RPi master `10.1.3.1/16`, RPi2 child `10.1.3.33/16`, RPi2 child usb0 `10.1.5.1/24`

**Routing Configuration**
1. In the RF24Gateway example working directory, create a file "routing.txt"
2. Add routing entries in a standard kind of format:
   ie: IP<space>NetMask<space>Gateway

Example routing.txt file:
```
10.1.5.0 255.255.255.0 10.1.3.33
10.1.4.0 255.255.255.0 10.1.3.34
0.0.0.0 0.0.0.0 10.10.3.33
```
First entry: Traffic for 10.1.5.x will use 10.1.3.33 as the gateway
2nd entry: Traffic for 10.1.4.x will use 10.1.3.34 as the gateway
3rd entry: Traffic not in the RF24Gateway network will be sent via 10.10.3.33 by default

**Additional Routing Configuration**
You can specify any combination of IP/Netmask in the routing.txt file to create unique subnets for routing traffic:
```
10.1.5.0 255.255.255.224 10.1.3.33
10.1.5.32 255.255.255.224 10.1.3.34
10.1.5.64 255.255.255.224 10.1.3.35
0.0.0.0 0.0.0.0 10.10.3.32
```
First entry: Traffic for 10.1.5.(1-30) will use 10.1.3.33 as the gateway
2nd entry: Traffic for 10.1.5.(33-62) will use 10.1.3.34 as the gateway
3rd entry: Traffic for 10.1.5.(65-94) will use 10.1.3.35 as the gateway
3rd entry: Traffic not in the RF24Gateway network will be sent via 10.10.3.32 by default

The general configuration remains the same, but you shouldn't need to add any static routes on the RPis via the linux routing table, just in the routing.txt file. As long as the ip/subnet combinations are correct, you should just need to have forwarding and nat enabled.

**Note:** The routing.txt file only needs to be present on the Master node (nodeID 0). Child nodes always send IP traffic to the master for lookup,routing,etc.

Hopefully I've got this info correct, so let me know if you have success with it.

SimonMerrett, 2020-07-09:
Great - I had a feeling a RF24 routing table would be required. You have implemented it in a really nice way - I will test and let you know how it goes. Thank you.

SimonMerrett, 2020-07-22:
@TMRh20 managed to do the first steps in testing this today. I initially got the `routing.txt` file location wrong (put it in the lowest level folder with `RF24Gateway_ncursesInt` -whoops. But also noticed that your example has it with a typo in the file extension `routing.txx`. With two nodes I can ping the tun endpoints and will check a third and fourth node tomorrow. Thanks again!

TMRh20, 2020-07-23:
Cool. Since I populated the routing.txx file with info, the file extension was named txx intentionally to prevent it from being loaded by default. I included a readme with info on that.

FYI there are some issues with internal routing when communicating directly from child -> grandchild or grandchild -> child nodes with TCP/IP that I'm looking at improving, but for the most part things seem to be working ok. Hope to improve this soon.

SimonMerrett, 2020-07-23:
> I included a readme with info on that.

Doh! Seen now. Should only nodeID = 0 have this routing table or does each node need one? I wrote one for the second node but I don't know if it's necessary (as not loaded by default).

TMRh20, 2020-07-24:
The non-master nodes all use 'dummy routing' for IP traffic currently, that is they just send all packets to the master node and let the master node figure it out, so no routing table except on master.

This causes the routing problems I mentioned, since traffic from child -> grandchild ends up getting sent  from child -> master -> child > grandchild, it gets ugly. I've been testing having the non-master nodes do address lookups for IP traffic, which would fix this problem and allow them to use a routing table as well. It adds a bit of time for small, direct packets, but makes a big difference when routing becomes an issue. I'll probably be able to commit the changes soon.

SimonMerrett, 2020-07-24:
Thanks - that makes a lot of sense. I've been having issues with different size microSD cards and the cloning thereof, but I should be able to confirm at least 3 way routing and perhaps 4 tomorrow.

SimonMerrett, 2020-07-24:
So I have 4x nodes running and some strange behaviour. Going to make a quick note here in case it's obvious to you @TMRh20 before testing a couple more things and trying to document it properly before showing more detail. 
Symptom - "slaves" (have to be careful how we use that now - only using it here until an alternative is agreed, for clarity) can ping another adapter on NodeID[0] (e.g. usb0, wlan0) but they can't ping adapters on another slave node and NodeID[0] cannot ping their non-tun_nrf24 adapters either. 

Could it be something to do with 
`sudo sysctl -w net.ipv4.ip_forward=1 # enable packet forwarding`
`sudo iptables -t nat -A POSTROUTING -j MASQUERADE #enable NAT` 

Perhaps NAT is now a problem?

TMRh20, 2020-07-24:
Sounds like the master node is missing it’s routing table, or it’s incorrect. It should typically be located with the RF24Gateway binary, in its working directory.

SimonMerrett, 2020-07-24:
> 1. In the RF24Gateway example working directory, create a file "routing.txt"

I am rubbish at reading your docs. I took this to mean `routing.txt` should be located in `RF24Gateway/examples`, rather than e.g. `examples/ncursesInt` (might have been distracted because that's where `routing.txx` is sensibly located (because it can apply to any example). Thanks for the clarification

So I have four nodes with three endpoints all pingable. Lovely work, thanks. I can't ping the ubuntu PC interface from the Win 10 PC attached to nodeID[0] but I can ping it from the other Pis and another PC. Pretty sure the routing table on the PC is fine but my track record so far has not been great.

> I've been testing having the non-master nodes do address lookups for IP traffic, which would fix this problem and allow them to use a routing table as well. It adds a bit of time for small, direct [...]

I don't want to keep an issue open longer than necessary but is it worth waiting for that commit and testing it before closing?

TMRh20, 2020-07-24:
Nope, if its all working for you so far, this issue can be closed. The additional stuff will work fine when I get it in. ;)

closed 2020-07-24 by SimonMerrett