nRF24/RF24Gateway #16: Not clear how to adjust MAX_PAYLOAD_SIZE to avoid 117 byte MTU problem

newfoundlandplucky, 2021-01-21:
Tried directions from https://github.com/nRF24/RF24Gateway/issues/8 but resulted in Linux core on Raspberry Pi Zero. Resolution of 117 byte max MTU RF24Network restriction appears out of date or no longer applies. Please advise.

There is a work around to instead adjust TUN MTU as described below. Do you recommend the workaround or instead modifying MTU RF24Network restriction?

```
RF24Gateway Ncurses Interface by TMRh20 - 2015
─────────────────────────────────────────────────────────────
tun_nrf24 IP: 10.10.3.1
<snip>
─────────────────────────────────────────────────────────────
 Program received signal SIGSEGV, Segmentation fault.
```

```
(gdb) where
#0  0xb6fbc030 in memcpy () from /usr/lib/arm-linux-gnueabihf/libarmmem-v6l.so
#1  0xb6ef9100 in RF24Gateway::handleRadioIn() ()
   from /usr/local/lib/librf24gateway.so.1
#2  0xb6ef8f08 in RF24Gateway::update(bool) ()
   from /usr/local/lib/librf24gateway.so.1
#3  0x00014364 in main ()
(gdb)
```

RF24Network_config.h
` #define MAX_PAYLOAD_SIZE  1490
`

Default tunnel MTU is left at 1500 when testing PAYLOAD 1490. Value suggested in original problem was also tried without success. Changing the Linux tunnel MTU value, even to 144 when PAYLOAD is 1490, appears to make no difference. Core is produced.

NOTE: Setting linux MTU on the tunnel to 144 allows the tunnel to function as expected when default PAYLOAD is 144.

Test environment is two Rasperry Pi zero modules with usb networking adaptors, RS232 serial console, and rLN24L01+ modules soldered to a Rasperry Pi hat perfboard. BCM and Linux NRF24 drivers both core dump.

TMRh20, 2021-01-21:
Did you recompile everything, including the RF24Gateway example after changing the max payload size?

newfoundlandplucky, 2021-01-21:
You clearly wrote "recompile the RF24Network, RF24Mesh and Gateway layers" in https://github.com/nRF24/RF24Gateway/issues/8#issuecomment-425710252. I didn't. Compiling all the modules fixed the issue. Thank you for responding so quickly.

closed 2021-01-28 by TMRh20