nRF24/RF24Gateway #19: RF24Gateway doesn't seem to work with a recent version of the libraries and Raspbian

microtronics, 2021-07-09:
I just did a fresh install on an up-to-date Raspberry Pi 2 and the Ethernet examples don't seem to work. The RPi has been installed using the script from http://tmrh20.github.io/RF24Installer/RPi/install.sh . Up to the RF24Mesh layer everything is fine and the mesh examples work as expected.

The setup is a RPi2 with RF24Gateway and a STM32 as Arduino with any of the RF24Ethernet examples. 

It doesn't matter if a server or client example is ran and different IPs have also been tried. Using the default 10.10.2.2 on the RPi and 10.10.2.4 on the examples. If a server example is used "curl" times out and "nc" will never show anything. On a client example the client will never succesfully connect.

While trying to connect RF24Gateway_ncurses shows that the RX packet count increases, but TX is not counting.
On a server example when attempting to connect the TX count increases, but RX does nothing.

```
sudo sysctl -w net.ipv4.ip_forward=1
sudo iptables -t nat -A POSTROUTING -j MASQUERADE
```

have been executed on the Pi. Ping also does not work, though the Pi can ping itself on both interfaces.

Client example log:
```
Disconnect. Waiting for disconnect timeout
354343 stop(), data: NULL
connecting
359346 stop(), data: NULL
359418 RF24Ethernet Network Write OK
359976 RF24Ethernet Network Write OK
361030 RF24Ethernet Network Write OK
363076 RF24Ethernet Network Write OK
connection failed
```

traceroute is not giving any hints:
```
pi@rf24gateway:~/rf24libs/RF24Gateway/examples $ sudo traceroute -I 10.10.2.4
traceroute to 10.10.2.4 (10.10.2.4), 30 hops max, 60 byte packets
 1  * * *
 2  * * *
 3  * * *
 4  * * *
 5  * * *
 6  * * *
 7  * * *
 8  * * *
 9  * * *
10  * * *
11  * * *
12  * * *
13  * * *
14  * * *
15  * * *
16  * * *
17  * * *
18  * * *
19  * * *
20  * * *
21  * * *
22  * * *
23  * * *
24  * * *
25  * * *
26  * * *
27  * * *
28  * * *
29  * * *
30  * * *
```

microtronics, 2021-07-09:
Using the SLIP_gateway the same issue appears to be present. I can see on the SLIP_serial that data is incoming from the clientExample and the RX counter goes up, however ping or connecting with the client/serverExample from the Pi don't work, Am I missing something?

TMRh20, 2021-07-10:
Unfortunately I’m not sure what to suggest. It sounds like you’re doing everything right so it should just work. You could try running an rf24 mesh example and see if the user payloads increases in the ncurses example. That would tell u if the mesh is working on the RPi.

microtronics, 2021-07-10:
Added another Node to let it run the mesh example and it works fine. The master (RPi) receives the payload and can also reply with no errors. It looks like it is Raspberry Pi related as only the tunneling/routing appears to be an issue. Have you tried it on a recent version of raspberry pi os lately?

microtronics, 2021-07-10:
Interrupt version also does not seem to work correctly out of the box...

microtronics, 2021-07-10:
The whole system works when using the Arduino Nano instead of the STM32. Could it be an endianess issue? What parts of the RF24-package assume a certain endianess?

TMRh20, 2021-07-10:
Yes I’ve recently run rf24 on a rpi4 with RPi OS. It seems odd that stm32 would work with every layer except RF24Ethernet. The uip stack assumes a certain endianness.

microtronics, 2021-07-10:
Changing to UIP_BIG_ENDIAN doesn't change anything and I wouldn't expect it, too, because STM32 are usually configured little endian anyways. It's still strange why it won't work , because I can see the whole connect message arrive at the STM32... The mesh works fine, so it is probably a RF24Ethernet issue with the STM32

TMRh20, 2021-07-10:
I don’t have an stm32 to test with so probably not much I can do with this. :/

microtronics, 2021-07-10:
uIP uses some crazy casting (4 byte ip will be stored as a two entry ui16_t array) and this appears to differ between an atmega arduino and stm32

microtronics, 2021-07-10:
@TMRh20 I opened the issue in the right repository

closed 2021-07-10 by microtronics