nRF24/RF24Gateway #41: RPI and TUN

epsi1on, 2025-01-09:
Hi,
I would like to make a network link of two RPI via rf24.
i did successfully ran the `/RF24/examples_linux/gettingstarted` on both devices, both can send and receive data successfully.

To setup network link, this is what i did ran:

Machine 1 (IP: 10.11.2.2/24)
```
sudo ip tuntap add dev tun_nrf24 mode tun user pi multi_queue
sudo ifconfig tun_nrf24 10.11.2.2/24

sudo sysctl -w net.ipv4.ip_forward=1
sudo iptables -t nat -A POSTROUTING -j MASQUERADE
```

Machine 2 (IP: 10.11.2.3/24)
```
sudo ip tuntap add dev tun_nrf24 mode tun user pi multi_queue
sudo ifconfig tun_nrf24 10.11.2.3/24

sudo sysctl -w net.ipv4.ip_forward=1
sudo iptables -t nat -A POSTROUTING -j MASQUERADE
```

after running the `RF24Gateway_ncurses` file on both PIs, this is result:
machine 1:

[image]

machine 2:

[image]

But when i try to `ping 10.11.2.3` from machine 1, there is no response from machine 2. I definitely think i'm missing something but cannot figure it out. :) Could you please tell me what am i missing?

Thanks in advance

TMRh20, 2025-01-09:
First off you probably don’t need to call the sysctl or iptables functions. Those are mainly needed for routing internal rf24 network traffic to external sources. You may want to keep.

Second you will need to choose a master node. All other nodes need to be assigned nodeIDs. See line 99 of the ncurses example.
‘uint8_t nodeID =22;’
You will want to match this to the last octet of the nodes IP address.  So let’s say 10.11.2.2 is master. You want to set the nodeid on 10.11.2.3 to 3 on line 99, Then on line 100 call ‘gw.begin(nodeID);’

epsi1on, 2025-01-10:
Kindly thanks for the information.
It did solve the problem, now i can ping machines via NRF links.
I did not tested yet, but do you think that using UDP will increase the speed for transferring audio?
Thanks again for great library.

TMRh20, 2025-01-11:
Thanks @epsi1on 

Yes I think UDP would be faster. I typically get speeds up to ~20KBps, but 10 is probably a good average.

closed 2025-01-12 by epsi1on

TMRh20, 2025-02-06:
Yeah, we can definitely include some more info in the docs.

I think we can give this piece of documentation its own page and link to it from the main page.

One item it think you may want to change is the pin number section, this could be a link to https://nrf24.github.io/RF24/ where we have a chart of the pins to use for RPi.

epsi1on, 2025-03-16:
Can the same technique be used for more than two RPI nodes. say make a IP network of three RPis?

Thanks

TMRh20, 2025-03-16:
Yes, the system utilizes RF24Mesh for addressing etc, so the networking part of it is basically automatic. You just need one "master" node and the rest can be assigned "nodeIDs" like you did with the first node. 

Keep in mind, with the mesh and multiple nodes, bandwidth can be more limited because nodes will "attach" to other nodes to form a connection if they are unable to "attach" to the master node directly. Nodes also need a bit of bandwidth to verify and renew their connection to the mesh, so you may need to manage bandwidth usage.